FrSky-OS/HORUS-X10-X10S-X12S-bug-report #234: Horus x10s power on button problem

fredde1982, 2019-03-12:
The radio is frsky x10s. I am running firmware version 1.3.03. I have the sd card in the radio. Its seems to be hard to turn on the radio with the card in. It seems to be that you need to hold in the button an very exact time to get it to start up, witch makes it difficult. If remove the sd card then it starts up like it should, no problem. It seems to be a strange behavior??????

Adelaxie, 2019-03-21:
Have you tried another sd card? sounds strange. we only has reports that radio can not open when some card(may bad? not sure) inserted.

fredde1982, 2019-03-21:
Hi
The card is correctly inserted. I can read from it when the radio starts up. I will buy a new card today and see what will be the result will be.

bob187, 2019-03-23:
HI i Also just got a x10 yesterday and after updating the firmware to the current i have the same issue as freddie. the button will power blue a split sec then turn red and stay red. u play with it a bit and it does the blue red blue and powers the screen. seems not to happen with no sd card in it

fredde1982, 2019-03-23:
Hi again
I bought a new card. I sadly did not test it with fros. I had the same problem with earlier firmware. The card was a class 10 / 32 gb, format fat32. 

What i did was to install open tx instead on the new sd card , now everything works fine.