FrSky-OS/HORUS-X10-X10S-X12S-bug-report #237: Horus x10s failsafe problem

fredde1982, 2019-03-24:
Hi
I have a problem what i think is related to the x10s. Iam using it with rxsr reciver. What happens is randome failsafe. Today i crashed because of it. The lowest rssi value was 50. The failsafe happend at 20 m in open terain, the radio did not say telemetry lost and the rssi was problally much higher than 50. 

Iam running the latest firmware for both reciver and trasmiter. I have replaced the rxsr once becuse i thought this was the problem. Did not help. The atenas is mounted with 90 degres between them.

The x10s i brand new. 

What is wrong iam getting very frustrated.

brainbubblersbest, 2019-06-09:
Hello, we are Experienced same issues as You.
Triggered by incompatible RX and TX Firmware Versions on a X-LITE and R-XSR
Try using Firmwares from the same publishing Date
We also flashed from LBT to FCC. The Listen before Talk Causes Dropped Frames and RX-Loss Flag on SBUS Protokoll. Lot more with FRSKY D16 and Telemetry as with D8 without Tel.
Never had such Failsafes happen again. The R-XSR raplacement helped nothing at all

Adelaxie, 2019-06-17:
@brainbubblersbest Can you tell which version did you use and now what version you are using?I want to make a comparison test.
Thanks in advance.

brainbubblersbest, 2019-06-18:
Hello Adelaxie, Issue with RXLOSS Flag in SBUS Protokol, 
happened with latest EULBT FW for the old X-Lite (over a Year old Firmware now)
and latest EULBT R-XSR Firmware 19.03.11

Bind with FRSKY X Mode (D16wTele)

Failsafes, Failsafes everywhere, we are tried 4 or 5 different R-XSR 2New ones and ones from Friends Quads, where we can be for sure that these are not faulty

Failsafes are triggered by the SBUS Specific RXLOSS Flag.

Flashed all the stuff to the FCC Version Firmwares (same Version Numbers) and had never an Failsafe again for 6 weeks now.
But now we are ilegal.

fredde1982, 2019-06-19:
Hi again
I was also running the lbt firmware. Never tried the none eu version. I tried sevral diffrent rxsr reciver same problem. I updated to latest firmware on both reciver rxsr and transmitter x10s.  Does not help, Randome failsafe. 

I have friend that has the same problem on his xlite with rxsr (lbt).

What i did to fix it was to give up and went over to crossfire.

brainbubblersbest, 2019-07-01:
Is there no intention from frsky to get rid with this