FrSky-OS/HORUS-X10-X10S-X12S-bug-report #238: Flaps- no Flap2 in output 1.6.1

Boroal, 2019-04-04:
In both airplane and glider setup there is no Flap2 in the output map. Therefore you can’t allocate a channel number to Flap2  Which means it isn’t possible to set ail-cmbflp as it uses flp1 and flp2 as does brk-cmb.

MIKE-DAILEY, 2019-04-04:
I do not have 1.6.01 available to test, but FLP2, FLP3 and FLP4 is in 1.6.03 and can be selected in outputs. 

Mike

choochoo2, 2019-04-04:
Perhaps the confusion is the OUTPUT MAP screen pages.  FLP defaults to ch8 which appears on the first page.  The user must advance to the second page to see FLP2 on ch9, FLP3 on ch12 and FLP4 on ch13.  All of these are the default channels and may be reassigned in this menu to the user's choice of channels.

MIKE-DAILEY, 2019-04-04:
That could be it.

Boroal, 2019-04-05:
My apologies it is the INPUT MAP where Flap2 is missing NOT the output map.

Boroal, 2019-04-05:
Updated to 1.6.03 still the same

choochoo2, 2019-04-05:
FLP2 is a split or clone of FLP in the same manner as AIL2, ELE2, and RUD2.  They are automatically operated by the control assigned to the primary function and it is only possible to assign a separate control through the use of an ex-mixer, which shouldn't be necessary.  If you need a flap function operated by a separate control, use FLP3.  The control assigned to FLP3 will operate FLP3 and its split or clone, FLP4.  The attached diagram is not perfect but it shows this relationship (It's more accurate to show the splits before the preset mixes but I haven't had a chance to fix it)
[Function flow diagram.pdf](https://github.com/FrSky-OS/HORUS-X10-X10S-X12S-bug-report/files/3049270/Function.flow.diagram.pdf)

Boroal, 2019-04-06:
Thanks that helps think I can now do what I needed to

closed 2019-04-06 by Boroal