FrSky-OS/HORUS-X10-X10S-X12S-bug-report #239: Horus X12S with FROS hang  Version is latest 1.6.01

scvhorus12, 2019-04-08:
hello serviceteam,
what happened. I changed modell - fine - I flew - fine - diconnect battery from modell and tried to check recorded telemetry data. System hang completely - no button nor switch worked. 
I switched off due holding on/off button longer as  normal. 
That makes me nervous. What's if it happens during flight. 
Any idea what's going wrong ?
br siggi

choochoo2, 2019-04-09:
#213 and #221 are somewhat similar incidents.  Thankfully all have occurred while navigating menus and not in flight.

scvhorus12, 2019-04-09:
Hello serviceteam, first of all. It might be similar but not same effect. In other cases old FW has been used. And that this issue happened more than once doesn't help. A fix would help. So set status to critical bug for that. You know what will happen if it occurs due flight ! If you are not willing to fix it - TX can't be used anymore.

choochoo2, 2019-04-09:
You may have mistaken me for a FrSky representative.  I am a user like you who has had a similar experience, #213.  In your case the system hung where in the other cases it powered off and/or rebooted.  Those failures seem like they could plausibly have a common cause.  FrSky seems to be of the opinion that this problem has been fixed with recent firmware.  It's difficult to prove that something so intermittent isn't failing but I am skeptical that it's fixed, and your incident feeds that skepticism.  I made the earlier comment mainly to link similar incidents in the hope this will improve the chances of getting a reliable fix.  

The fact that all three of these documented incidents occurred while the user was working with the menus and not in flight may not just be good fortune but perhaps an emerging pattern that could assist in troubleshooting.

Your last post suggests you believe you were using the most current firmware.  The current X12 releases are 1.6.03 and 1.6.04 beta.

scvhorus12, 2019-04-09:
Hello ,
first sorry for missunderstanding - don't want to be snippy up to you.  And you are right - it's not latest version.  Ok I will install 1.6.3 as soon as possible. 
BTW - checked download area - version 1.6.01 isn't longer available. That's strange isn't it.

choochoo2, 2019-04-09:
V 1.6.01 on the X12 and 1.3.01 on the X10 were an interesting example of how not to manage firmware versions.  First they posted .01 beta.  This had a couple of minor flaws but basically worked pretty well.  Then they took down that beta and posted .01 as a release version.  My assumption (and I think most users) was that since .01 beta had proven useful, it was being released and a new beta would soon appear.  In fact, the release version of .01 was a previously unpublished version.  It corrected one or two of the minor problems in the beta but introduced several new bugs that were much more annoying.  Since it had been published as a release version and not a beta, many users loaded it, including those who don’t usually load betas.  Since the previous .01 beta had been removed from the website, this left many people with no way to return to a better version.  Much unhappiness prevailed until .03 was released.  .03 seems to contain the best of both versions of .01.  I honestly haven’t looked much at .04.  I expect .01 is not on the website because FrSky would like to put this fiasco behind them.

 

It should be noted that the bugs mentioned above were all annoyances, not crash risks.  Yours was the first incident I’ve noticed of the system crashing since the previous firmware generation (1.5.x on the X12).  FrSky had previously noted on both incidents #213 and #221, which were earlier generations, that they believed it had been corrected.  I’m not an engineer but to me a hang, reboot, and spontaneous power down, are all consequences of a system crash.  It’s certainly possible the causes were different but I suspect there is an underlying flaw that has yet to be isolated and is likely responsible for all of these incidents.  This has got to be really hard to troubleshoot, given the rarity of the incidents and that no one has been able to replicate it.  It’s also hard to prove with confidence that it has been fixed.

 

Pardon my rambling, I could very well be all wet on this.  Please keep that in mind.

 

Rick

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus