FrSky-OS/HORUS-X10-X10S-X12S-bug-report #240: Dist-GPS does not compute range w/X10S

thock1960, 2019-04-11:
When using the GPS (GPS V2) sensor with a RX6R receiver and a X10S transmitter there is no calculated data for the Dist-GPS telemetry value.  Also in configuring this field in the settings, the units selection is only meters, it seems that ft, miles or Km would also be desirable too.

From the manual for the X109s) it says, "If the transmitter contains no sensor the distance will be measured between the place where the GPS unit first achieved GPS lock after switch on, and the on-board sensor.", so I assume this feature should work with the X10(s), similar as to how the GPS altitude is calculated.

Adelaxie, 2019-05-16:
We will test on it. thanks for report/

WismyYao, 2019-11-13:
fixed

closed 2019-11-13 by WismyYao