FrSky-OS/HORUS-X10-X10S-X12S-bug-report #241: Horus x12s - Fw: 1.6.04 - Battery voltage values tx shown incorrect wt 3 cells 18650...

Lombardoa, 2019-05-12:
On my Horus x12s I replaced the NI-MH battery pack with three 18650 li-ion cells. The radio works normally, while the value read on transmitter battery voltage its wrong ... I have seen that the values above 10.7V are wrong. While below this value the values read are correct.
Please, can you verify?
In advance ...

Lombardoa, 2019-05-19:
Hello Adela...
why not reproduced...you can remove the battery , then check with a power supply at 12V...easy!!!

Adelaxie, 2019-05-20:
yes, we have good display here.. how is the value wrong?pic?

Lombardoa, 2019-05-21:
Hi Adela,
well, I'm sending you 6 photos where you can see the voltage problem. In 
the right corner, below, you can see the time.
Photo n.1, starting condition, 11.7V
Photo n.2, after about 25min. 11.3V
Photo n.3, after about 55min, 11.2V
*Photo n.4, after about 1 hour, 11.8V ... here the voltage rises, and I 
don't know why ... this is the real problem.*
Photo No. 5, the voltage continues to decrease normally, then stop 
everything and go to sleep.
In Photo No. 6, you will find all the info about the firmware.

I advise you to remove the battery, power the tx with a variable power 
supply. Set it to 13V, then turn on the radio, calibrate the value read 
on the horus, and then start decreasing the voltage from 13V to 6V.
Try it several times ... if you find inaccuracies, please fix everything.
I thank you for your patience and cooperation.

Arturo

Lombardoa, 2019-05-21:
[image]
[image]
[image]
[image]
[image]
[image]

Adelaxie, 2019-05-23:
testing

Adelaxie, 2019-05-27:
@Lombardoa Hi,sorry, we have been testing on this for days bu we still cannot reproduce what you reported, can you check with the real voltage  by removing the battery out and measuring by voltagemeter?

Lombardoa, 2019-05-30:
Hi Adela,
as you suggested, I measured the voltage simultaneously with two multimeters. One on the output pins of the external module, Ph4.
And the other, on the battery output connector. To increase the battery discharge, I enabled internal RF and also mp3 music. Below are the results.
1) Start with Ph1 (12V - time 19:13)
2) after a while Ph2 (11.6V - time 20:00)
3) after a while Ph3 (12.2V - Time 20:07) (here the value rises)
At this point the problem comes from the balancing circuit of the new LI-Ion batteries. 
The case is solved.
Thank you again for your cooperation and patience ... see you soon ... Arturo

[image]
[image]
[image]
[image]

Adelaxie, 2019-05-31:
Thanks for your information and I will close this issue then.

closed 2019-05-31 by Adelaxie