FrSky-OS/HORUS-X10-X10S-X12S-bug-report #243: Neuron parameters not shown on default telemetry screens

choochoo2, 2019-05-30:
Even if everything is working properly in a Neuron setup, none of the telemetry values are shown on any of the eight pages of telemetry screens.  They may only be displayed by using the TELE-SETUP and Screen-Position functions to assign the values to one of the positions on the first six pages.  The need to do this isn't intuitively obvious to new Neuron users and doesn't seem to be described anywhere.  This confusion is aggravated by the fact that a new user doesn't necessarily know what parameter names they are looking for.

Parameters already exist pre-Neuron for the various values.  Voltage-fas, RPM, temp, etc. exist on the default telemetry screens as they could be supplied by other sensors.  A new user is likely to be expecting the Neuron to supply the pack voltage, for example, by populating the Voltage-fas field, as would occur with a current sensor.  I certainly was expecting this, so the new user doesn't even know he should be looking for escVolt, and they won't come across it on the default telemetry screens because it isn't there.

The Neuron instruction sheet should clearly list the names of the values the Neuron supplies, and not just in the section related to OTX.  The name should be listed as they apply to FrOS, especially since it appears the FrOS and OTX names differ.  Further, they should appear in the default telemetry screens.

Depending on your design philosophy (Do you intend all possible parameters to be shown on the default screens, or only 24 values plus the cell and RB screens?)  If you intend all values to be shown, two screens should be added to display the 7 Neuron parameters.  If you don't want to add screens, some less used values should be replaced on the default screens with at least a couple of the Neuron values so they will appear where they will be found by users searching for them.  Also, if only 24 values will be shown and the user must pick and choose, that should be made clear somewhere.

Adelaxie, 2019-05-30:
Thanks for your suggestion, we will modify this in instruction and next FrOS release.
By the way, we will start to put hands on FrOS improvements .
Any good suggestions will be very appreciated.

miff69, 2019-05-30:
There's also an issue with the Neuron telemetry values for logical switches. The values are shown, but they have wrong units. e.g. escV is in xxxxx mV (I assume).

JohnMeul, 2019-05-31:
Hello Adela:
I wonder if you are aware that the Gas Suite cannot display its telemetry
on the Horus with FrOS. Correspondence with Andrea_Huang over at RCGroups
is aware of the problem. The Gas Suite has been out for over a year now,
and many of us are hanging on to them waiting for the fix to be implemented.
I was wondering if it was on your fix list?
Thanks

Adelaxie, 2019-05-31:
New parameters will need to make changes on structure, we will do it in a new version.

JohnMeul, 2019-05-31:
Ok. So what do we do with the ones we bought and never worked?

choochoo2, 2019-06-01:
Since you asked for suggestions on FrOS improvements, how about:

- Create a script/menu for setting up a Neuron from the tx in a manner similar to the SXR stabilizers.
- More logic switches.
- Create a logic switch test for "between" two values.
- Make all functions available in the RATE/EXPO menu instead of only a select few.  This kind of arbitrary design restriction is one of the reasons I moved from Spektrum.  You don't know how the AUX and other functions will be used so don't presume RATES/EXPO won't be needed.  As an example, your own SXR stabilizers need rate and offset settings on the gain channel to make a remote gain knob work properly.
- Allow the option to use the output of a function as the SRC for an ex-mixer instead of the current situation of using only the input control of a function.  This would allow cascading ex-mixers and some other capabilities not currently present.  This could be implemented by allowing the SRC pull down list to select either a control or a function.  If a control is selected it works like now, if a function is selected the function output is used.  Some conversion of existing ex-mixers would be necessary on updating.

Adelaxie, 2019-06-03:
@JohnMeul
It means new model file structure in FrOS.

JohnMeul, 2019-06-03:
Thank you Adela.

The current FrOS version is 1.3.03.
FrOS version 1.3.04 is in Beta stage.
Version 1.6.xx is being worked on.
It looks like version 2.xx is almost underway as well.

Are any of the above FrOS versions addressing the Gas Suite issue?

Thanks.
Best Regards,
JohnM

Adelaxie, 2019-06-05:
V1.6.xx is for Horus X12S and v1.3.xx is for x10/x10s/
There is lua file now for v1.3.xx to show the parameters.
And I also updated it in Gas Suite download section. please note.

JohnMeul, 2019-06-10:
I did not realize that The FrOS versions were different for the X10 and
X12. Thank you for sharing that.

All the other details you listed have already been addressed by me. They
are all known, applied, and explained already, although in reviewing this
thread, I see I have not explained it all here.

Here is a summary of the communication that I (and others) have had with
Andrea Luang at RCGroups:

- As previously noted, Lua script or not, The Gas Suite does not perform as
a Smart Port telemetry device is supposed to.

 - All other Smart Port devices report to the telemetry screen on the
radio. This is the established performance expectation of Smart
    Port devices as established by FrSky.

- The Gas Suite system does not and cannot do this. Therefore it is not a
properly functioning device.

- The lua script function is not a solution. It is useless while in flight.
   It takes a minimum of 10 to 15 seconds of concentration on the radio
screen to get it running, and if you leave the lua script page
    after getting it running, you have to do it all again.

- That is not how the Gas Suite is supposed to work, and certainly not how
it is implied to work by FrSky in their advertising.

- To make the Gas Suite work requires a change to the FrOS data structure
(answered by you or another FrSky tech,
    I can't remember which, sorry!)

So onward to the one remaining question:
Would it be possible for you to tell us if there is a FrOS version
currently in development that will correct this problem, and can you tell
us approximately when it will be released?

Thanks.

WismyYao, 2019-06-12:
Thanks for your attention to FROS.

We are modifing the system architecture to fix the problem that telemetry data cannot be added and  improving the existing system.  It's a big job, and it will take some time.

JohnMeul, 2019-06-13:
Hello WismyYao:
Thank you so much for your reply.
I am glad to know that you are working on fixing the problem.

I guess I must assume that   "...and it will take some time", means that a
solution is not going to be available in the foreseeable future.

I am very pleased with the FrSky products I have used to date, but I feel
like I was blind-sided but the Gas Suite.

I am very disappointed that FrSky would release a product that was only an
idea, but was marketed as if it actually worked.

However, I seem to be the only one that feels that way, so I guess that's
it.

Thanks again for your response.

Best Regards,
John M.

Maverick440, 2019-06-26:
regarding the original comment by choochoo2 about the Neuron ESC and the fact that none of the telemetry values are shown on any of the eight pages of telemetry screens.
I to have this problem and am currently using FrOS version 1.2.05 on my X10S
I see the FrOS version 1.3.03 offers support for Neuron ESC, does this meen that the telemetery will now work on Horus X10S if its running 1.3.03?

choochoo2, 2019-06-26:
Maverick,
Please read my response to this in the third place you posted this question:
https://www.rcgroups.com/forums/showpost.php?p=42176219&postcount=301