FrSky-OS/HORUS-X10-X10S-X12S-bug-report #245: EX-Mixer Suggestions

choochoo2, 2019-06-06:
While creating the EX-Mixer Tutorial and later trying to help a couple of people online, I came to learn that the current ex-mixer structure has some limitations.  Some of this was mentioned already in issue #243 but it seems like this deserves being an issue in itself with expanded explanation.  Here are suggestions for each screen in the ex-mixer section to expand and clarify their capabilities:

**Summary screen:**
Number the ex-mixers to make it easier to edit or document a model where several are used.  Display them on the summary screen in number order.  Offer a choice in the pop-up menu to reset ALL ex-mixers, with a confirmation required.

**Main screen:**
This is the most important suggestion.  The current structure of an ex-mixer uses a control as its source input.  This is masked somewhat by the way the source is specified by selecting the function associated with the control, but it is the control, not the function that works as the SRC input now.  If AIL is the SRC in ex-mixer3 with AUX as MIX-TO, the real source is not AIL but J1 (the AIL control) as is illustrated by the red line in the diagram.  This works surprisingly well for many situations but is limited and tends to cause some confusion as seen in issue #244.  The functionality of ex-mixers would be extended if it were an option to select the AIL function output as illustrated by the green line in the attached diagram.
[image]
From a user interface point of view this would be easy to implement.  In the SRC drop down list, include both controls and functions.  If the user selects a control, the ex-mixer would work as it does now, using that control as source.  If a function is selected, the ex-mixer would work as proposed and illustrated by the green line above.  Among other things, this structure would permit cascading mixers.  When this change is implemented and a user updates, it may be necessary to convert existing ex-mixers to specify the control as input so existing mixers continue to work as written.  Not certain but some could break if this is not done.

A minor suggestion; instead of MIX-TO, use DEST as a label.

**Weights screen:**
Permit weights up to 200%.

**Offsets screen:**
Add the ability to specify a trim control.  Put in a drop-down list where the user can select a trim or knob (or any control?) and a method selection, either + or * to choose whether the mix result is added or multiplied by the trim.

**Curves screen:**
Add some check boxes to specify the selected curve is to be flipped vertically, horizontally, or both.  This can be helpful in situations where symmetrical but opposite curves are needed so the same curve can be used for both.

WismyYao, 2019-06-12:
Thanks a lot.
I'll take your suggestion into serious consideration.

choochoo2, 2019-06-12:
Certainly

Burgerman-X10express, 2020-04-14:
I cant get more than 1 exmixer? Is that a bug?

choochoo2, 2020-04-14:
Ex-mixers are selected by rotating the scroll wheel.  If no mixers are set up the selection may be difficult to see, however.  As you scroll through the 10 mixers, the only thing that changes is the position indicator on the right (blue arrow in screenshot).  Clicking will activate the selected mixer with a pop-up choice to edit or reset.  As mixers are populated (used) it becomes easier to see the selection change.  At least this is how it works on systems prior to Express.  I don't have a Express to test the newer firmware.  If it isn't working as described on your tx, perhaps there is a bug.

If you haven't done so, you might want to read through the Ex-mixer Tutorial from my blog page here: https://www.rcgroups.com/forums/member.php?u=354219

[image]