FrSky-OS/HORUS-X10-X10S-X12S-bug-report #246: Requesting a Standard Way to Flash A Horus to a Fresh Clean Firmware State

brainbubblersbest, 2019-06-09:
Hello, i Flashed my Horus x12s a dozend Times now.
after each Flash i noticed a different Issue.
Especially for the "ERROR SD_Card Horus internal Firmware EU"

I know this has to do with not compatible Files in the Internal Flash Drive.

like Models are stored in Internal Flash that are been created in EU Firmware, 
and after Flash to NEU this circumstance trigger this error.
Sometimes you need to Erase the Transmitterfile and so on...

I searched weeks now for a Instruction to a working Standard method, like a Production Firmware reset.

I have absolut no Idea, if its possible to reach a clean flashed Transmitter Software state, ONLY 
with the Horus flash Contents
AND the latest Firmware ZIP like 1.6.0.4 

Are there any previous Version Firmware Bins or Files in Folders that need to be Flashed and copied in a specific Order, to reach an actual Clean newish 1.6.0.4 Version State?

An Backup of a Clean Internal and SD Card Flash Content on the FRSKY Download Site, would making Things so much easier.

Thank you.

Adelaxie, 2019-06-17:
@brainbubblersbest The flash contents are included in the firmware update packages. (no in some very old version updateing),please check and if anything else please post here.

brainbubblersbest, 2019-06-18:
Hello, Please imagine this aktual state of the Transmitter:
-Flashed  a douzend times
-also flashed from fcc to eulbt and back
-now have corrupted Internal or external Flash (have no IDEA why or what in the specific)
-have no Backup and will Build the TX Flash Internal and External from Scratch. (like a Virgin)

How would this be done, if you cant Trust Files on your TX and will make the TX Virgin like

Adelaxie, 2019-06-19:
so, what would you like to suggest to do on this? Hope we can help to resolve your troubles.

brainbubblersbest, 2019-06-19:
I will like to know, if its enough to clean a corupt horus data system in THIS way?
-Plug Horus to PC in USB Storage Mode
-Erase both Drives, Internal Flash and SD Card
-Then Copy **Flash Contents-Horus X12S** from https://www.frsky-rc.com/wp-content/uploads/Downloads/FLASH-X12S.zip to Internal Flash from Horus.
-Copy Folders Included in Firmware Folder from 1.6.04 Beta from https://www.frsky-rc.com/wp-content/uploads/Downloads/Firmware/X12S/FW-X12S-1.6.04-Beta.zip
-Copy Firmware X12S_XXXXX_1604_Beta0219_frtx.bin into internalflash:/Firmware/
-Erase Transmitterfile 
-Switch Horus off after Safe remove USB and Turn on with pressed Clickwheel to start Flash
-Follow instructions Calibrate Battery and Sticks

Is this the right way to do it and will this work?
The Manual in the Download Section is Unprecise, 
I will like to know if i do anything wrong in the steps above.

A Firmware Image flashable  with a Windows Tool would make a lot things easier for the Customers.