FrSky-OS/HORUS-X10-X10S-X12S-bug-report #247: Gas cut off  problem, wrong usage or what else ?

robertobuti, 2019-06-14:
My be possible i'm wrong in usage, but until now i don't find any info about this problem, so i hope someone can help me.

trim & subtrim = 0
Gas (ch3)
limit +110, -115 according with servo mechanical limits
travel +100 (full throttle), -50 (minimum rpm) according with engine gas requirements

Cut off menù, enabled by H switch
cut off = -110 (position required for throttle full closed )
Trigger value not used at the moment.

Seem the cut off doens' work , after many tentative i understad the cut off value must be within "travel" values, i'm expect within "limit" value, in this way is impossibile calibrate the gas engine & have cut off working.

I'm wrong ? where the problem ?

**General information**
 - Hardware: X12S EU14
 - Version 1.6.04 beta 02/28

MIKE-DAILEY, 2019-06-14:
@Adela this feature was working correctly, but some where along the builds it does not seem to work like it did anymore.  Or I've always used it on the low side of the throttle movement around -100. 

The work around seems to be for the above situation is to change the -50 and move it to the +side of the throttle movement.  Throttle movement will be -100 to +50.  Then the throttle cut can work down around the -100 position.  

Here is how I made it work:

1. Move throttle trim and set in the trim center.  When the other settings are done the center should be normal idle speed than can be adjusted for faster or slower idle using the trim.  Trim will move from the -100 center to -75 full high trim and full low trim -125 
2. Adjust - end point to -100 and + end point to +50.  Set low limit to -125 and default high at 110.  
2. Adjust the carburetor linkage so -100 is the normal low idle and +50 is full throttle. 
4. In throttle cut, adjust TRI POS to -75 and SF POS to -125.  -75 is the set point where throttle cut can be used. With the throttle position faster than -75 the cut can not be accidently triggered.      

Depending on the total movement needed for the throttle cut adjust the SF POS and the end point - limit.

Mike

robertobuti, 2019-06-17:
Thank Mike, i try with your setting, but the problem still the same.
No way to obtain a valid excursion of servo when cut_off is activated. I'm expecting to have the possibility to have cut-off excursion from minimum throttle like (-50 or -70) to -125 or something like this, seem not possibile. Excursion of servo when cut off is activated if always very limited, See the movie at following link,  [img]https://i.imgur.com/Gkk8Fyg.mp4[/img] in the movie before the full servo gas escursion and then the the cut-off excursion (too limited).

MIKE-DAILEY, 2019-06-17:
If the idle position shows -100 or close to -100 on the monitor screen throttle cut moves the channel to -125.  It that what you are getting?  From -100 to -125 is a lot of movement. 

Mike

robertobuti, 2019-06-17:
Mike honestly i don't agree, from -100 to -125 is small excursion, about 5% of full excursion and not good to cut off the engine. (the excursion is visible on movie of previous thread), in any case why is not possible set a bigger cut off excursion for cut off?

MIKE-DAILEY, 2019-06-17:
Did not say it was the final solution.  I said it was a "work around".

Mike

FrSky-GX, 2019-06-18:
we have reproduced it , we will check what happend and fix it.

FrSky-GX, 2019-08-29:
There is another way to set it, keep the THR channel travel -100--+100 and limit -110--+110,  set the THR curve weight 75 offset 25, set THR CUT as you did, then it shoud work.