FrSky-OS/HORUS-X10-X10S-X12S-bug-report #248: Horus x12s - FROS  1.6.04 LBT - Slow flight mode problem...

Lombardoa, 2019-06-16:
As above, when i put some value on Slow variable and try, i no have no one result on servo.  Same  problem on heli program, Plane program and so on. Before update it worked fine. Please. can you check... thanks in advance...A. Lombardo

[image]

[image]

FrSky-GX, 2019-06-18:
we haven't reproduced this problem , could you make a vedio with details?

Lombardoa, 2019-06-19:
I can make a video... but i can't put it here...only pictures...

Lombardoa, 2019-06-19:
As I have already written, with  a airplane program, whatever value I put, the speed of the servos does not change. I don't see any delay. While the voice or sound that I associated with the interested key, SWB_DN, with the play track function, results with a delay, the audio is bad for the whole time relative to the value inserted on SLOW. After this delay the audio works well.
Hope you can reproduce...

Lombardoa, 2019-06-25:
[FrSky X12S Horus User Guide 20180928 - page 46.pdf](https://github.com/FrSky-OS/HORUS-X10-X10S-X12S-bug-report/files/3324346/FrSky.X12S.Horus.User.Guide.20180928.-.page.46.pdf)

Adela, please ... on page 46 of the manual it is written that if the value written in SLOW MODE is used to avoid fast drives, such as flaps, gears, ailerons. As I have already explained to you, in my case, with the AIRPLANE program, it does not work. If I want, a delay, for example on FLAPS, I have to use the SPEED function, described on page 30.
While, it should also work in SLOW MODE ... Why doesn't it happen? ... Am I doing something wrong? Thanks for the support...

closed 2019-06-25 by Lombardoa