FrSky-OS/HORUS-X10-X10S-X12S-bug-report #250: Enhancement Suggestions for FrOS

choochoo2, 2019-08-01:
**Misc. Suggestions:**

•	There should be an “All” choice for flight modes wherever they appear in menus.  That is, instead of just specifying that the menu settings apply to “Landing” mode, for example, there should be a choice on the list to apply the settings to all flight modes. For convenience include this as a check box in ex-mixers as well.  For instance while “N” is the first box for ex-mixer modes, “A” might be the first box instead, where selecting “A” sets or clears all the other boxes.

•	LS and RS should be selectable as 3-pos switches in switch lists.  This would greatly enhance their usability for things like selecting flight modes, 3-pos flaps, flight controller settings, the SXR stabilizer modes, etc.  Implementation should be straightforward, they should appear on the switch list as up-mid-down just like other switches.  They should be available everywhere switch selections appear.  Selecting them as switches in one place should NOT disable their use as pots in other places.

•	There should be a Special Function to announce (speak) the current flight mode when it is first selected, once, or periodically like other special functions.  It would be good to build in a small delay, perhaps ¼ to ½ sec to avoid announcing modes that are being skipped as the pilot moves switches.

•	Logic switches should include a logic “NOR” operation.

•	Logic switches should include “between” and “not between” operations.  Perhaps labeled “Bet” and “Not Bet”, or by >< and <>.

•	Logic switches should include flight mode as a condition option.

•	There should be more logic switches.

•	Consider implementing named global variables.

•	All menus that have multiple screens should cycle back to the first screen when PAGE DOWN is selected on the last screen.  Currently some do and some don’t.

•	In the Glider type RATES/EXPO menu, rates are selectable only by flight mode.  While this makes sense sometimes, other times it is quite inconvenient.  It would be good to include an option to specify switch positions here instead.  Perhaps a switch selection field could be included with a default of null where selecting a switch position disables the flight mode selection field, or perhaps there could be separate settings for switch positions within each FM.

•	The telemetry log should record values using the same units the pilot selected for display of each field.

**Pitch compensation:**

In general, elevator compensation in FrOS is less convenient and less intuitive than it should be.  Elevator compensation should be included in menus that have a pitch effect requiring compensation.

In the Aircraft type the FLAP SET menu should handle all that is needed for setting up flaps on most typical aircraft.  Currently it doesn’t seem to be working but even if it were, it lacks elevator compensation.  This menu should include settings for ELE and ELE2 and there is room to do so on the existing screen.

In the Glider type it’s more complicated.  Probably the FLAP SET menu won’t be used but if it is to be useful it should also include settings for offsets for all surfaces, as the Aircraft type does, and it should include ELE and ELE2.

The BUTTERFLY, CAMBER MIX, and MOTOR CURVE menus should include full settings for elevator compensation including weight, offset, and curve.  This might also be included in the FLAP SET menu.  A better approach might be to create a new menu for PITCH COMP with the curve settings and selection field to link to Butterfly, Motor Curve, Camber, and Flap functions.

WismyYao, 2019-08-12:
Thanks for you suggestions.

choochoo2, 2019-08-16:
I would like to illustrate the usefulness of some of these suggestions by offering the following example:

While writing the Glider Guider’s Guide I wanted to setup three flight modes controlled by LS and announced audibly when each is activated.  The original objective was five modes controlled by LS and RS but it quickly became apparent that there aren’t enough logic switches to do this so I settled for three.  To implement this required 3 custom sound files, 3 special functions, and 4 logic switches utilized in this manner:
[image]

LSW1 used to activate and announce Launch mode when LS is high.

LSW2 used to activate and announce Landing mode when LS is low.

LSW5 checks if Launch or Landing are active (LSW1 or LSW2).

LSW6 is simply NOT LSW5 and is used only to announce Cruise mode.  (Cruise mode doesn’t need a switch to activate as it is activated by default if Launch or Landing are not active)

Improvement # 1:  If the logic switches had a “NOR” logic test, LSW6 could be combined into LSW5, eliminating the need for one logic switch.  This could also be accomplished with logic switch tests for “between” and “not between” which have also been suggested and would reduce the number of logic switches needed in many situations.

Improvement # 2:  If there were a Special Function to announce the current flight mode, the need for LSW5 would be eliminated.  This would also eliminate the need for the custom sound files and two of the three special functions used.

Improvement # 3:  If LS and RS (or all pots) could be selected as 3-pos switches, the need for LSW1 and LSW2 would be eliminated.

By making the suggested improvements, instead of consuming half the eight provided logic switches, this rather simple objective of selecting and announcing three flight modes with LS could be done without using any logic switches, without any custom sound files, and using only one special function.  This approach would also be much simpler and more intuitive to implement.  The original objective with five flight modes would be possible, although it would require a few logic switches.  The suggested improvements would apply to other objectives and make the whole system simpler and more versatile.

GUT83, 2020-06-12:
Improvement on the vario telemetry  : 
- add a range min/max where to remove the tone...
- ad the possibility to adjust the sound