FrSky-OS/HORUS-X10-X10S-X12S-bug-report #251: FLAP SET Bugs

choochoo2, 2019-08-01:
The FLAP SET menu has multiple bugs.

When the menu is first entered it appears like this where a switch can be assigned to settings # 1.
[image]
If you change to the # 2 screen and back to screen # 1, it now appears like this, where the switch assignment is greyed out.
[image]

Worse, it appears the # 1 offset value of 50 used in the examples applies always, regardless of the switch position and the # 2 settings.  In other words, the menu is unusable.

The fact that these bugs are just now being reported suggests this menu isn't being used.  This would make sense given that its current design isn't very useful.  Please see issue #250 for suggestions on how to make this menu more useful.

WismyYao, 2019-08-12:
Thanks for your bug report .

WismyYao, 2019-11-13:
fixed

closed 2019-11-13 by WismyYao