FrSky-OS/HORUS-X10-X10S-X12S-bug-report #252: Logic Switch "NOT" operation displays extraneous data

choochoo2, 2019-08-01:
This isn't a particularly important bug but you might want to know about it anyway.

Unlike the other LOGIC SWITCH operations, the NOT operation uses only one source and the data in the second source is meaningless.  Here is an example logic switch using NOT.  The data in the second source fields are greyed out.

[image]

Although the greyed out data is meaningless, it still displays on the Logic Switch summary screen...

[image]
… and it displays on the Logic Switch status screen:

[image]

The circled data fields should be blank on these last two screens.

WismyYao, 2019-08-12:
Thanks for your bug report.

WismyYao, 2019-11-13:
Fixed in X12S 1.6.06 / X10(s) 1.3.06

closed 2019-11-13 by WismyYao