FrSky-OS/HORUS-X10-X10S-X12S-bug-report #254: Neuron BEC Lua Script not working

choochoo2, 2019-08-02:
The Lua script  provided for setting the BEC voltage in v1.3.04 does not seem to function.  The script runs and fills the display but the data fields do not populate and it isn't possible to set the voltage.

This could be part of a more general problem as the S6R Lua script doesn't seem to work either.  It also fills the screen but does not populate the data fields.  The S6R settings under the System menu seem to work fine, however, so there is another way to setup the S6R but no way to set the Neuron BEC voltage using FrOS.

These are the only Luas I've tried on the X10 FrOS and neither work as intended.

Adelaxie, 2019-08-09:
[fros_lua.zip](https://github.com/FrSky-OS/HORUS-X10-X10S-X12S-bug-report/files/3484653/fros_lua.zip)
Please try with the lua files.

choochoo2, 2019-08-09:
I tried the Lua files from your zip above.  The sbec script produced exactly the same result as before, did not populate the fields, even after a minute of waiting.  To illustrate this I went to the special functions menu and created a print-screen option and returned to the Lua script to take a screenshot.  This time the fields populated after just a few seconds, before a screenshot could be captured!  

The script worked fine to adjust the voltage.  Using the original script from the firmware download now also worked the same, although it did not before.  Changing to another model and back, the script still works fine.  

I don't know what to take from this.  The script you posted above seems to be the same as the older one.  At first try neither worked, now they both do.  I've no idea what changed, do you have a thought on this?

The SXR script, however, still does not work.  Neither the new one or the old one, both still fail to populate the fields.  This is less of an issue for me as the SXR menu works so the Lua isn't needed.

choochoo2, 2019-12-21:
It looks like the nature of this problem should be changed from "Neuron BEC Lua Script not working" to "Neuron BEC Lua Script flaky".  It does work sometimes but apparently never the first time the script is run.  Usually if the script is exited and re-started it will work but sometimes it enters a state where only rebooting the Neuron will allow it to work.  This was tested with 1.3.06.

choochoo2, 2020-01-18:
Some additional information.  Firmware 1.3.06.

In order to work, it must be beyond swamping range and the following sequence is necessary:

*Turn on the TX
*Plug in the aircraft
*Wait for tx to receive esc telemetry
*Run Lua script, it will fail
*Exit and restart Lua, it will succeed

If you exit and restart the script again it will be fine but if you restart the tx the script will fail and stay failed unless you also reboot the aircraft and use the sequence above.  All of this is entirely repeatable, see video (please pardon the dust):

[[url]https://youtu.be/cKDn7igd2N8[/url]](url)

PS:  If you were paying close attention you may have noticed another firmware problem.  The SWITCH WARNING is bogus, no switches are out of position as indicated by the white color.  It seems I am the only lucky user to encounter this problem, as reported in issue #225.