FrSky-OS/HORUS-X10-X10S-X12S-bug-report #256: Neuron Telemetry Bugs

choochoo2, 2019-08-03:
I hate to keep complaining about the Neuron but the more contact I have with it the more bugs are apparent.  There are problems all over the place, at least in FrOS.  Ironically the product seems to be better supported by the third party OTX system than by FrSky's developers.  I just did a final setup and flew with a Neuron for the first time yesterday and found:

- None of the telemetry parameters can be calibrated, and some need it.  My escVOLTS, for example, reads a quarter volt high.  This is potentially dangerous as 1/4 volt is significant near the end of the pack.  That flight ended with only 10% left in the pack while the telemetry was suggesting there was more.  We need the ability to calibrate telemetry values.

- The voice readout of escVOLTS ignores the decimal point.  For example, 11.35V will read out as "one thousand one hundred thirty five volts".  The pilot has to convert that to useful information in their head while flying.

- The voice readout of escAMPS ignores the decimal point as well and reads out in a manner similar to the above.

- NONE of the esc telemetry parameters appear in the log file.  This is a serious shortcoming that greatly undermines the usefulness of the Neuron's sensor capability.  You can't keep track of all these values in flight, even with voice readouts.  We need the ability to analyze the log data to evaluate the power system performance.  That is what users expect and is the main reason for buying a Neuron in the first place.

WismyYao, 2019-08-12:
Thanks for your bug report.

Maverick440, 2020-06-03:
I fly electric planes and use the voice telemetery as a guide wheist flying. 
if you set escAmps to generate a voice comand at say 30 Amps the voice comand states 3000 Amps, yet the telemetery readout on the Horus screen showes 30 Amps!!
This is at best anoying and sheds doubt as the the acuracy of the rest of the telemetery setting.
It also apers that the same can be said for escRPM as the voice comands do not match with the RPM being shown on the screen