FrSky-OS/HORUS-X10-X10S-X12S-bug-report #257: Incorrect value shown on LSW page - Horus x12s - FROS - Fw:1.6.04 beta lbt

Lombardoa, 2019-08-07:
I describe a small problem with LSW functions
As shown in ph1, I entered the value of 4.7V on LSW4, to monitor the voltage of the RXbattery.
While on the first page (ph2) it shows me a value of 4.6V.
Thanks for your support.

[image]

[image]

Adelaxie, 2019-08-09:
its caused by losing of precision during data conversion.

Lombardoa, 2019-08-10:
Hi Adela,
thanks for your answer...
I remember that i no have this losing of precision with old fw 1.5.05 LBT...
It's possible improve this precision?...
Thank again...

A.Lombardo

closed 2019-08-14 by Lombardoa