FrSky-OS/HORUS-X10-X10S-X12S-bug-report #259: X10 Express glider menù buttons not shown

alex998r, 2019-10-07:
Buttons on page 3 and 4 of the Glider type are not show, no letters on them, but they seems to functions. Some other buttons, like input/output map, are correctly shown but doesn't functions

X10 Express version 2.0,02
I also copied the CONFIG folder into FLASH.
If you want I can send some pictures
Regards
Alessandro

alex998r, 2019-10-07:
[image]
[image]
[image]

Burgerman-X10express, 2020-03-01:
Absolutely the same here in every detail.

Royalevo, 2020-04-13:
Hi, i programmed a glider today, with my new express radio, and all buttons on page 2-3 is not working (from input map) I chose Vtail .
Grx8 in acct mode 

Someone have same issue?

Regards
Geir

GUT83, 2020-06-11:
Hello, recently buyed a lbt Horux X10s express running the 2.0.02_Beta fros, especially because it has FrOS instead of opentx, first model......glider menu is a mess, no binding working, no updtae of the receivers through the radio (update aborts systematically)...I tried the 2.0.03 FrOS from the module download page, the glider mode works corretly, BUT no gimbals recognised, no switches, nothing except the beutiful color screen......pffff, is it serious?????

i flashed to the OPENTX last version, and everything works normaly, binding in access mode, updates etc....but for open tx there are better radios that are really less expensive!
ashamed, and really disappointed