FrSky-OS/HORUS-X10-X10S-X12S-bug-report #263: Copy in Model Select fails with <10 characters in name

thock1960, 2019-11-20:
In the systems menu to copy a new model from a current model, if the name of the original model is 11 or more characters, the copy function does not work.  The message shown is  "The name length will be longer than can be set!" and will not copy the message. However the total length of the field is 15 characters, thus if all 15 characters were used I could see the reason for this message.
However, the system should allow the copy of model with a name that has characters up to 14 or 15 in length.

Steps to reproduce the behavior:
1. Go to 'SYS'
2. Click on 'Model Select'
3. Scroll down to a model to copy and press the Enter, choose 'copy' Enter.
4. message is  "The name length will be longer than can be set!"

**Expected behavior**
With a name of 14 (or 15)  characters, the message should be, "Are you sure you want to copy this model file ?"

**General information**
 - Hardware: X10S
 - Firmware Version: 1.205

Thanks for your consideration,
Terry

WismyYao, 2019-11-27:
The max characters of model name is 15.  When copy a model,  a '_' will add before model name . If the length of your model name is 15,  it will be 16 after copying , over the max name length .

thock1960, 2019-12-02:
The issues is that after the message of "The name length will be longer than can be set!"  and you select OK, nothing happens, the model is not copied. It also seems this message should only occur if the model name is 14/15 characters in length, but it occurs when the model name is 11 or greater in number of characters.  The only way to copy a model is to ensure the model name length is not over 10 characters. Please let me know if you need more details.