FrSky-OS/HORUS-X10-X10S-X12S-bug-report #264: x10s express/Software 2.0.01/ SxR setup menus not working

1Gump, 2019-12-28:
**Describe the bug**
Have been trying for days now to use the x10s express system/SxR setup menus to set up an S8R. Menus do not show nor allow any options to select (only show -- or blank values). Only graphics are on screen 3/3 of the s6r label up. Screen 3/3 does nothing when "enter" is pressed.
.
The S8R binds and servos operate as normal. channel 10 switch causes correct responses (AL,Stab,Off). 
I can operate the servos with the sticks UNTIL I enter the FROS SXR setup screen - the servos don't respond to the sticks after that, but respond to the gyro rotation. Only rebind allows control of servos with sticks again.
.
no prior setup to this Rx (PC, STK) was performed.
.

**To Reproduce**
Steps to reproduce the behavior:
1. Bind S8R (D16, channels 1-16, telem 1-8)
2. Turn on system. wait until Rx blue light stops blinking - turns green. check for servo control - affirm. Move Rx to check for gyro response - affirm.
3. Scroll to System/SXR setup menu on x10s express FROS.
4. page 1/3 ... "wing type" box = ---   "Mount type" box = ---    no graphics. selecting yields no choices.
5. page 2/3.. all selections --- or blank. no choices allowed.
6. page 3/3 (described above). graphics but no response to "enter"

**Expected behavior**
This screen should populate with choices to setup the S8R.

**General information**
 - Hardware: x10s express. brand new. 1st model/Rx
 - Version Tx
Software 2.0.01
Phys id. 23
Int module. Isrm-s-x10/neu
Version. 1.1.0/1.1.3
"s6rconfig.cfg" is in the flash config folder.
S8R successfully flashed with latest firmware S8R_FCC_20190531.frk
SD card included.

**Additional context**
i am assuming this SXR setup screen(s) is(are) behaving as if the Rx was not an SxR or turned on with RF link. However, it is a brand new S8R, it is operating gyro compensation (tested by moving Rx by hand) AND servo control by sticks when this setup is attempted.

Burgerman-X10express, 2020-02-28:
Just to add. I am seeing the exact same thing. The feilds are all empty. All my transmitters are FCC and latest firmware etc. 

Seperate but maybe related issue?
I updated my 3 SxR recievers (S6R and S8R) none of them would bind. So I had to go back to the old firmware on those.

Burgerman-X10express, 2020-03-01:
Been doing some more digging around.
Two things.
1. Right at the top of the file in flash: config/S6Rconfig.cfg when opened as a text file theres a line break directly after the first character. So you get a "{"  with nothing after it...
Delete the white space. That may or may not be required.  I am no programmer.
And add the two files SXR.lua and SXR.luac to either the internal flash (in any folder it seems) or on an SD card. Navigate to it, in system/files, and choose activate or run or whatever it says. And it then works. So while we wait a year, at least theres a way to use the TX I just bought a week ago. Fun no?  This is one of a few issues. Range, and random servo movements, etc are another. 

We wont mention the confusing firmware and what works with what thing.  I spent a week trying to get everything working. And now I am in the wrong "area". I used to think that FRSKY were great. And flexible. And gave US the power. Now we have this fixed area problem, (I work in the US and live in the UK, so need to swap at will!). Cant keep getting a dealer to swap things around every few months. Thats rediculous. Thats why I bought into this system. So I had a little control.

Burgerman-X10express, 2020-03-01:
Just tested. Run as a script from sd or flash. Works normally. Theres around 50 or so settings, adjustments and configurations and it calibrates and sorts out its sensors and saves everything wirelessely as it should. 

But the menu is still screwed up. Needs fixing ASAP!

Burgerman-X10express, 2020-04-05:
Also no way to set up the R9 Stab reciever from the TX as the SxR lua script does not work at all here. Express TX at least. Actually I cannot find a lua script for that rx at all.

isidresa, 2020-04-10:
Also not working in V2.0.02

isidresa, 2020-04-10:
Also not work  in FW V2.0.02

isidresa, 2020-04-10:
Also not working in FW 2.0.02

Burgerman-X10express, 2020-04-11:
Yes I updated to V2.0.02. and still nothing. Express.

Royalevo, 2020-04-13:
I have the exact same problems. In addition the SW freeze in several ocations when making new/copying and editing models.

Hope they are working really hard on new SW, and also have som QA on it, or else I have to switch brand. This is just not good enough for a 6000NOK radio

Regards Geir
Norway

Burgerman-X10express, 2020-04-14:
So summary:

The menu on horus X10 and X10 Express does not work on the S6R, S8R, or R9 STAB. All feilds remain blank.

And if you run the SxR script, from the SD card - then it works, but not on the R9 STAB.

This appears to be regardless of firmware on reciever or any EU, NON EU mode and also ACCESS on the R9 Module and STAB reciever.