FrSky-OS/HORUS-X10-X10S-X12S-bug-report #265: x10s FrOS powered off during the flight

hyepilot, 2020-01-20:
Yesterday I updated x10s firmware to latest  version 1.3.06 which was release  on 01.16.2020. Today during the flight spontaneously the power turned off resulting in total loss of the plane. I have been using x10s with no issues for two years.

WismyYao, 2020-01-20:
Sorry to hear that. 
Could you share me more details ? Such as model configurations, telemetry logs and so on.

hyepilot, 2020-01-20:
Here are more details.
The model configuration:
Hangar 9 Ki43 Oscar with Saito Fg 90r3 engine. Two  XR8pro receivers connected to RB20. Two LiPo batteries. The firmware of both receivers was updated the same day.
The plane had many flight with the same configuration with old firmware. 

Unfortunately I was not recording telemetry. 

About 3min into flight the transmitter powered off. It was fully charged.

choochoo2, 2020-01-20:
May be related to #213, #221, #239.

WismyYao, 2020-01-21:
> Here are more details. The model configuration: Hangar 9 Ki43 Oscar with Saito Fg 90r3 engine. Two XR8pro receivers connected to RB20. Two LiPo batteries. The firmware of both receivers was updated [...]

Two  XR8pro receivers   ？  Is R-X8Rpro ?

hyepilot, 2020-01-21:
Yes, the two receivers are RX8R Pro. sorry for the typo.

WismyYao, 2020-01-21:
Which mode do you use ? EU or NEU ? @hyepilot

hyepilot, 2020-01-21:
NEU

WismyYao, 2020-01-22:
We found that the radio will turn off if  the battery does not connect well  . Could you help us do a test ? Shake the battery or the radio when the radio is turned on.

hyepilot, 2020-01-22:
I did the test you requested. I shook the radio for few minutes, the power
did not turn off. I also opened the back and checked the battery. It is
connected properly.

WismyYao, 2020-02-10:
Is this the same with you ?
https://github.com/FrSky-OS/HORUS-X10-X10S-X12S-bug-report/issues/267

hyepilot, 2020-02-10:
I don’t have Vari-h, so I can’t tell. My radio shut down after few minutes of flight.  

 

#267 <https://github.com/FrSky-OS/HORUS-X10-X10S-X12S-bug-report/issues/267>

Burgerman-X10express, 2020-05-01:
I could make the horus express turn off by shaking the TX. Those battery tray spring loaded connectors are as dangerous today as they were in the 90s. Fitted in all the cheap transmitters full of AA cells... They only worked when you squeezed the TX! 

I just fitted 2x LG M50 21700 5000mAh and SOLDERED the wires as I always have for the last 40 years of flying. Thats another potential crash reason removed. You need the "thing verse" new battery base 3d printing to fit those. But that makes it plug and play, doubles run time too. 

Thats not to say there is not still a bug in the code doing this. Just another possible crash cause eradicated. You may be wise to do the same.