FrSky-OS/HORUS-X10-X10S-X12S-bug-report #266: New Logging bug

choochoo2, 2020-02-01:
There is a problem with the way 1.3.06 is recording log files for this year.  Please look at the screen shots attached.  The file showing the date as 2019-13-23 was recorded on 2020-1-23, so the recorded filename is effectively the 23rd day of the 13th month of 2019.  Three such samples are attached.  The current date and time display correctly on the X10 screen.

Windows cannot read the dates correctly either as shown in the screenshot from windows file explorer which shows the date modified as blank.

The last file is puzzling.  I believe this was made on 1/17/2020 but the system date and time in the X10 have always had some trouble with years.  A few weeks ago I updated to the correct system date and time.  I'm not certain exactly when I did that.  I suspect (but do not know for sure) that this file was recorded before I made that change and the date was off a couple of days so recorded as the 19th.  Regardless of the exact date; for this file only, the year was recorded correctly but the month shows as 00.

[image]
[image]
[image]
[image]
[image]

WismyYao, 2020-02-10:
Thanks for your report .

choochoo2, 2020-04-01:
Yesterday (March 31) I flew and tried to view the log file but couldn't find it.  Tried a couple of other models to test logging, still couldn't find the day's file.  Then I remembered the problem above and went looking for a file with a date Windows can't read for the 15th month of 2019.  Sure enough, there's the day's log file.

Just checking, the system date was Jan 29, 2019 and the time was a couple hours off.  How and when the date got set back to that date I have no idea, it was set correctly when the post above was written.  How the system month of 1-2019 translated into a log date of 15-2019 also makes no sense.

Today (April 1) I reset the system to the correct date and tried logging again.  Now it is working correctly and the log file is dated to 4-2020.  We'll see what happens next month.

Aside from this weird logging problem;  For maybe $2 at Walmart you can buy a kid's watch that can keep the correct time and date with reasonable accuracy.  I don't see why it's such a challenge for a $400 transmitter.

Here are some images showing yesterday's bogus file date:

[image]

[image]

[image]