FrSky-OS/HORUS-X10-X10S-X12S-bug-report #267: Horus X10s shutting down

Mark4247, 2020-02-08:
Updated Transmitter, Receivers and Vari-H on Feb. 2, 2020 with latest firmware. Now every time my Vari-H is plugged into the receiver (S6R), the transmitter completely shuts down. If I unplug the Vari-H the transmitter (so far) works fine. Plug in the Vari-H and within 60 seconds the transmitter will shut down. All worked fine before the recent updates

WismyYao, 2020-02-10:
Thanks for your report .
We will check it ,

WismyYao, 2020-02-24:
[1.3.07 Beta0224.zip](https://github.com/FrSky-OS/HORUS-X10-X10S-X12S-bug-report/files/4242870/1.3.07.Beta0224.zip)

Could you help me to test it ?

Mark4247, 2020-02-25:
Installed firmware today, so far the vari-H has not shut off the
transmitter. 🤞

Mark4247, 2020-03-13:
Finally got to fly yesterday, so far so good! No shut downs.