FrSky-OS/HORUS-X10-X10S-X12S-bug-report #268: Horus X12 and gas Suite no temp or flow data

alex998r, 2020-04-05:
Hello, I connected a Gas Suite to an RX8R Pro receiver, latest FCC FrOs firmware 2.1 on both radio and RX.
I have no Temp1, Temp2,Flow data on Telemetry display of my X12.
I tried resetting the telemetry but I receive only RSSI and RX battery voltage.
I aslso conncted Gas Suite to the usb adapter on my pc running freelink, it connects and I can also change some parameters, I can read temp and flow, so the Gas Suite is not broken. 
The discovery of new sensors is not present like the X9D radio, please do I have missed some step to register new sensor?
Thanks

JohnMeul, 2020-04-05:
Hi Alex998r:
I have the same system with an X10S,Rx8R and Gas Suite. I went through the
same problems and discussed it at length with FrSky and other users
approximately a year ago. The Horus FrOS ACCST system does not report Gas
Suite telemetry data to the telemetry screens. There is a non-resolvable
system incompatibility. The only option is to run a luascript file from
your tx, which will then report the data to a makeshift screen on your tx.
There is an entire thread discussing this issue on RCGroups.

https://www.rcgroups.com/forums/showthread.php?2922046-FrSky-Horus-X10-and-X10S-and-New-Express/page634

Good Luck...

alex998r, 2020-04-05:
Hello John, many thanks for your help. I already ran the lua script, the last found on FrOs 1.0.7 and it is ok, but it is not acceptable FrSky puts out a device that is not compatible with teir radios. Regards

JohnMeul, 2020-04-05:
I agree wholeheartedly Alex998r.