FrSky-OS/HORUS-X10-X10S-X12S-bug-report #282: FROS Horus x10s express Archer receivers channel assign issue

bitbot2010, 2020-08-20:
I recently purchased two R10 PRO archer receivers. I have Registered both with different UID's and binded successfully.
First Issue:
-When in options menu of either receiver, the PIN assign menus for both receivers only show pins 1-8 although they both have 10 pins for the 10 channels.
Second Issue:
-When in options menu supposedly I should be able to assign say channel 1-8 to receiver number 1 and 9-16 to receiver number 2 (I did set UID 1 for receiver 1 and UID 2 for receiver 2 when registering them). However the "pin menu" only shows channel 1 through 8 on both receivers.

 I have seen people successfully assigning different channels to different receivers on OPENTX however in FROS it doesnt seem to work!

seagovernor, 2020-08-28:
Same issue here.
I recently purchased Horus X10S Express (Ver 2.0.06 the latest version) eith ISRM (Ver 2.1.0) and two Rxs Archer R10 Pro and R8 Pro.
After successful firmware upgrade I could register these two Rxs and bind them. But Rx1 the R10 Pro with 10 pwm pins, but in the menu it show only 8 channels.

seagovernor, 2020-08-28:
Is it possible to use two mixed Rxs R10 Pro and R8 Pro in one model?
I need 12 PWM channels for two gas engine powered model, Thus I thought R10 Pro and R8 pro might be good for the option.
I have registered R10 Pro as UID 1 and R8 Pro as UID 2. The Map option for R10 shows only 8 pin map but actually pin 9 for channel 9 and pin 10 for channel 10 are working properly, but the R8 Pro (UID 2) could not be mapped to channel 9 to 16 (or channel 11 to 18), only psooiblw channel 1 to channel 8. Please teach me what I am doing wrong. Have to use same type of Rxs for the job, possbly two R8 pro?. R10 Pro has two battery connecters and gas engine safe, which I like.

bitbot2010, 2020-08-28:
It will not work until FrOs is updated and fixed, I switched to OpenTx and there you can easily assign pins to the right channels with no issues.